artisan-roaster-scope/artisan #1724: Yoctopuce meteo 2 is not working

beanoccio, 2024-10-22:
### Problem Description

Is it a bug in artisan, or is incomplete documentation the cause?
The device should automatically enter the ambient temperature, humidity and air pressure into the roasting profile.

[image]

### Artisan Version

3.0.3 Continuous Build

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

DIY Roaster Arduino Mega

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-10-22:
Does a direct USB connection work for you?

beanoccio, 2024-10-22:
What do you mean by a "direct" USB connection?
The yoctopuce meteo is plugged into a USB port and provides the data via a virtual hub.

beanoccio, 2024-10-22:
Ok, without virtual hub, it works.
Thanks a lot!

beanoccio, 2024-10-22:
Thank you very much. Now I can sleep well.

closed 2024-10-22 by beanoccio

MAKOMO, 2024-10-22:
Works for me also with the VirtualHub on macOS. There might be a firewall or something in between in your case.

MAKOMO, 2024-10-22:
Note, the VirtualHub way is to remove access the device.