artisan-roaster-scope/artisan #1725: Multiple Event

Robjones6146, 2024-10-23:
### Problem Description

Multiple Event does not work for Artisan Commands. According to the literature is is supposed to unless I have misunderstood.

> _Note that command actions can be sequenced by using a semicolon in-between single commands. Eg. the sequence
> pidmode(0); pidSV(180); PIDon; sleep(300); PIDoff
> is valid and configures the PID to manual mode with target temperature 180C and turns it on for 5 minutes. Note further that sequences of different action types can be combined using a Multiple Event event action that triggers (hidden) button actions, each grouping a list of potential different action type._

**To reproduce:** Multiple Event [ 11, sleep(3), 60, popup(Test Message One, 3), message(test message) ] >> 11 is activated, then nothing.
**This works:** Multiple Event [ sleep(5), 11 ] >> Sleeps for 5 seconds then 11 is activated
11 is a button (move slider to 65%)
60 is a button (move slider to 0%)

Not sure if this is a bug or never was a feature or I am doing something wrong, sleep() works, I assumed the other should work too.

message() and popup() are very useful during multi events to show completion of each step and delay until the next step.

### Artisan Version

3.0.2

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Lots of Phidgets

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-10-23:
Could you please attach a screen shot showing the details of your configuration, or even better, attach your Artisan settings as .aset.txt file?

Your example uses syntax which is not valid and it is unclear where you put that string exactly. THANKS!

MAKOMO, 2024-10-23:
Note that the description of Multiple Event Actions only allow for comma separated numbers and sleep commands. You cannot put there other Artisan commands like `popup` or `message`. A work around is to put those in extra hidden buttons and refer to them by button number.

closed 2024-10-23 by MAKOMO

Robjones6146, 2024-10-23:
Exact copy of Multiple Event:   11, mesage(test message), popup(Test
Message two, 3), sleep(3), 60, popup(Test Message two, 3)

Robjones6146, 2024-10-23:
Ok, thanks, is there a reason for that?

Robjones6146, 2024-10-23:
[image]

MAKOMO, 2024-10-23:
Yes. You can only use comma separated numbers and the sleep() statement within a Multiple Event command.

Robjones6146, 2024-10-23:
.... and NO spaces between commas!

Robjones6146, 2024-10-23:
Thank you for prompt reply and assistance. Much appreciated.