artisan-roaster-scope/artisan #1729: Preparation of all available statistical values ​​in a structured text or CSV file with auto-export

beanoccio, 2024-10-26:
Automatic saving is very useful, but the data is very artisan-specific and cannot be used further with the aid of a computer.
The statistical values ​​can be saved manually as a PDF or as a text file. Unfortunately, the information is poorly prepared for further processing in studies over several roasts.

I would like an automatic export of all available statistical values ​​as a txt file according to the following scheme:

ID;Name;Value;unit

Example:
1;Ambient temperature;35;°C
2;Ambient humidity;42;%
3;Ambient pressure;1023;hPa
4;Batch Size;0.4;g

Currently, I save the Statistic Box manually as a text file and read this text file into a database for further processing for comparative statistics of several roasts.
However, the unsorted text file statistics require an enormous amount of string operations to prepare the data for further processing.

roasterdave, 2024-10-26:
The Summary Statistics feature was not intended to be used this way.

For comparative analysis use the Ranking Reports.  The PDF and Web formats are formatted for humans.  The CSV and Excel reports include all the data you will need.  

Use File>> Report>> Ranking>> CSV... or Excel...   Select one or more profiles of interest, click OK, then choose the path and file name for the report.

closed 2024-10-26 by roasterdave

beanoccio, 2024-10-26:
Of course, I know the numerous export and evaluation options in Artisan.

However, the workflow is very cumbersome and the exports as CSV only contain the event information in time with the sampling rate in order to compare a profile. But the basic information about the properties of the roast is missing in these exports.

All these 2 dozen possible data that are provided in the "Statistics Box. Which export can provide this in an automated and structured way?

I couldn't find a CSV, TXT, or xml export that also includes the properties of the roast.

roasterdave, 2024-10-26:
>I couldn't find a CSV, TXT, or xml export that also includes the properties of the roast.

Did you try my suggetion above? All the roast property data will be found in the Ranking CSV and Excel reports.  
>Use File>> Report>> Ranking>> CSV... or Excel... Select one or more profiles of interest, click OK, then choose the path and file name for the report.

beanoccio, 2024-10-26:
What I mean is this data (yellow highlighted) in a better structured form as text or CSV file.

like:
ID; name; value; unit

This made it easier to read this information into a database.
If I do several roasts per day, then at the end of the day I have a lot of documents in a folder. I want to import this information into a database as a data set per batch.
But if this basic information could be saved automatically as a text file, then I can read it into a database very easily.
In the database I then automatically have the roasting information in a data set and the links to the batch profiles and the PDF report.
I need the data set with the roasting properties for comparisons and search functions.

[image]

beanoccio, 2024-10-26:
My workflow:

I have a selfmade database with the following timed workflow.

Roast batch -> off-gassing storage -> roasted bean storage -> post-blending -> filling -> product storage

A storage table for the off-gassing of the roasted beans (5 to 20 days after roasting).
A storage table for the roasted beans (ready for filling or delivery)
A storage table for the products (single origin or post-blends)
A storage table for bags and packaging and labeling.

With a simple query I can see how many roasts I have to be done and when in order to be able to deliver the desired products.

Unfortunately I always have to write the roasting properties manually from Artisan into the roast batch table in the database.

I also use Artisan + but unfortunately this solution does not include management of off-gassing storage, roasted bean storage, post-blending, storage of packaging material or filled products.

roasterdave, 2024-10-27:
The Roast Report in your screenshot was not intended to be parsed by machine.  It is meant for human eyes.  

All the roast property data you want and lots more is structured and contained in the CSV Ranking Report.  Create one Ranking Report at the end of the day.  It takes moments to create.  Select all the day's roasts and use the output CSV file as the source data for your database.

_NOTE: For the CSV Ranking Report be sure to use the [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) v3.0.3 or later.  An inconsistency between the columns created for CSV and Excel reports was corrected._

beanoccio, 2024-10-27:
Ok, now I have the last Continous build Artisan 3.0.3 (17b841e) 
I can now use the CSV or XLS data to update the database. The data is available in a structured form. Thank you very much!
Unfortunately, it is not possible to create this data automatically. The workflow is a bit cumbersome, but it is acceptable.