artisan-roaster-scope/artisan #1730: Adding Exponential Curve To Designer.

Unconvino, 2024-10-27:
The curve designer doesn't appear to support exponential curves.

 I use a quick calculation to get to my desired endpoint from wherever I am using an exponential curve (eg Newton's Law of Heating). 

I currently do the calcs externally during the roast as a decision aid but would like to be able to make the curve show on Artisan. Since it's an exponential curve its a simple matter to calculate the BT curve from the RoR and vice versa. 

I can do the maths for this feature if someone can work out how to port a simple exponential equation into the curve designer. I would prefer it to use only temperature and time since charge: I don't take account of FC or the existing "phases" in my roasting.

roasterdave, 2024-10-28:
Have you looked at the Plotter and Math tabs at menu Config>> Curves ?

Unconvino, 2024-11-09:
I have.

None of the available options does what I want. The exponent part only allows a parabolic or cubic fit. A parabolic curve corresponds to a linear RoR curve which is what I'm trying to avoid. 

The ln(x) seems to be just a curve fitting function with no way to manipulate it.

roasterdave, 2024-11-10:
Indeed, the Math tab is all about curve fitting.  

I pointed out the Plotter tab above because it allows you to write any equation you want.  Click the Help button to see the available variables and math functions.  The Plotter allows generating persistent curves from equations P1 and P2.  Hover the mouse over the Background and ET/BT buttons for more information.  

If you want to start from a curve fit, copy the equation from the Math tab into the Plotter tab and edit it there.

Unconvino, 2024-11-10:
Thanks, I will give that a go

Mark

MAKOMO, 2025-01-12:
Stall

closed 2025-01-12 by MAKOMO