artisan-roaster-scope/artisan #1732: Slider Command only controls the first command

Robjones6146, 2024-10-28:
### Problem Description

**The Slider is controlling the first command but not both.**

Event: H-C PID    Action: VOUT Command    Command:  out(0,{},687340:4);out(0,100-{},687340:3) 
[image]

Hot, Cold and Loft sliders work fine.
The H-C PID Slider only moves the Hot damper.

Please can someone tell me where I am going wrong.
- Removed all spaces.
- Tried 100-v as suggested then changed to 100-{}

### Artisan Version

3.0.2

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Lots of Phidgets

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-10-28:
Rob,

I just literally pasted your `VOUT Command` to a slider after a Factory Reset and turned debug logging on (click on the plus icon while holding the ALT (OPTION on mac) key and got the following:

```
10-28 14:30:59 Dummy-1    artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,46.0,687340:4)
10-28 14:30:59 Dummy-1    artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:4)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,75,None,False,False,687340,4)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,76,None,False,False,687340,4)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,77,None,False,False,687340,4)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,4,None,False,False,687340,4)
10-28 14:30:59 Dummy-1    artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,54.0,687340:3)
10-28 14:30:59 Dummy-1    artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:3)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,75,None,False,False,687340,3)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,76,None,False,False,687340,3)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,77,None,False,False,687340,3)
10-28 14:30:59 Dummy-1    artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetVoltageOutput,4,None,False,False,687340,3)
```

No matching Phidget was found thus the command failed (no Phidgets were connected), but you see the two VOUT commands being processed. The first (trying to) send `46.0` to the first Phidget and the second (trying to) send `100 - 46 = 54.0` to the other Phidget.

So maybe you can verify that each of those single commands (maybe if issued from a button instead of a slider) work as expected with your hardware. If not something might get wrong with the addressing or device attach. Note that if you have multiple modules connected overall power might develop into a problem as USB ports only deliver a restricted amount of power.

MAKOMO, 2024-10-28:
I think I can reproduce the issue using two channels of one OUT1100 PWM module. All seems to work fine if not specifying the modules serial number and HUB port. Once those are specified, only the first of the series of commands succeeds. I am investigating....

MAKOMO, 2024-10-28:
If possible, please attach your artisan.log (recorded in Debug mode) file as your issue might still be different to the one I just discovered due to different hardware (and I do not have access to the hardware you are using).

Hold ALT/CTRL keys (OPTION/COMMAND on macOS) while clicking the plus icon to toggle the debug mode and hold just the ALT key to have an email open with the log files attached.

THANKS!

Robjones6146, 2024-10-28:
Hope this is it.

MAKOMO, 2024-10-28:
Nothing attached. You need to rename artisan.log to end in .txt and drag-and-drop it here on the GitHub Web page into a comment field. An email replay removes any attachment.

Robjones6146, 2024-10-28:
[log_file.txt](https://github.com/user-attachments/files/17546806/log_file.txt)

MAKOMO, 2024-10-28:
No, this is somehow the encoded email. The logfile attachment is a pure text file. You find that log file on your computer on Windows in the following location as `artisan.log`.

C:\Users\<USER>\AppData\Local\artisan-scope\Artisan

Robjones6146, 2024-10-28:
Thanks for your patience.

10-28 17:03:42 MainThread artisanlib.main      DEBUG  : message: Event # 40 recorded at BT = 175.2CC Time = 03:12
10-28 17:03:42 Dummy-908  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.795,684949:4)
10-28 17:03:42 Dummy-908  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:03:47 MainThread artisanlib.main      DEBUG  : message: Event # 41 recorded at BT = 176.3CC Time = 03:17
10-28 17:03:47 Dummy-469  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.665,684949:4)
10-28 17:03:47 Dummy-469  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:03:53 MainThread artisanlib.main      DEBUG  : message: Event # 42 recorded at BT = 177.9CC Time = 03:23
10-28 17:03:53 Dummy-1408 artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,5.055,684949:4)
10-28 17:03:53 Dummy-1408 artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:03:59 MainThread artisanlib.main      DEBUG  : message: Event # 43 recorded at BT = 179.6CC Time = 03:29
10-28 17:03:59 Dummy-602  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,5.25,684949:4)
10-28 17:03:59 Dummy-602  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:04:00 MainThread artisanlib.canvas    DEBUG  : alarm 59 fired
10-28 17:04:00 Dummy-243  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.925,684949:4)
10-28 17:04:00 Dummy-243  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:04:00 MainThread artisanlib.main      DEBUG  : message: Event # 44 recorded at BT = 179.6CC Time = 03:30
10-28 17:04:00 MainThread artisanlib.main      DEBUG  : message: Alarm 59 triggered
10-28 17:04:03 MainThread artisanlib.canvas    DEBUG  : alarm 11 fired
10-28 17:04:03 MainThread artisanlib.canvas    DEBUG  : alarm 27 fired
10-28 17:04:03 MainThread artisanlib.main      DEBUG  : message: Event # 45 recorded at BT = 180.3CC Time = 03:33
10-28 17:04:03 MainThread artisanlib.main      DEBUG  : message: Alarm 27 triggered
10-28 17:04:03 MainThread artisanlib.main      DEBUG  : message: Alarm 11 triggered
10-28 17:04:06 MainThread artisanlib.main      DEBUG  : message: Event # 46 recorded at BT = 180.8CC Time = 03:36
10-28 17:04:06 Dummy-422  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,5.055,684949:4)
10-28 17:04:06 Dummy-422  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:04:12 MainThread artisanlib.canvas    DEBUG  : alarm 41 fired
10-28 17:04:12 Dummy-544  artisanlib.main      ERROR  : [WinError 2] The system cannot find the file specified
Traceback (most recent call last):
  File "artisanlib\main.py", line 10312, in call_prog_with_args
  File "subprocess.py", line 1026, in __init__
  File "subprocess.py", line 1538, in _execute_child
FileNotFoundError: [WinError 2] The system cannot find the file specified
10-28 17:04:12 MainThread artisanlib.main      DEBUG  : message: Event # 47 recorded at BT = 182.1CC Time = 03:42
10-28 17:04:12 MainThread artisanlib.main      DEBUG  : message: Alarm 41 triggered
10-28 17:04:14 MainThread artisanlib.main      DEBUG  : message: Event # 48 recorded at BT = 182.5CC Time = 03:43
10-28 17:04:14 Dummy-919  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.795,684949:4)
10-28 17:04:14 Dummy-919  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:04:15 MainThread artisanlib.main      DEBUG  : message: Event # 49 recorded at BT = 182.9CC Time = 03:45
10-28 17:04:15 Dummy-1429 artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,7.789999999999999,687340:4)
10-28 17:04:15 Dummy-1429 artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:4)
10-28 17:04:18 MainThread artisanlib.main      DEBUG  : message: Event # 50 recorded at BT = 183.5CC Time = 03:48
10-28 17:04:18 Dummy-396  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,1.49,687340:3)
10-28 17:04:18 Dummy-396  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:3)
10-28 17:04:23 Dummy-1366 artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,5.81,687340:3)
10-28 17:04:23 Dummy-1366 artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:3)
10-28 17:04:23 MainThread artisanlib.main      DEBUG  : message: Event # 51 recorded at BT = 184.4CC Time = 03:52
10-28 17:04:30 MainThread artisanlib.canvas    DEBUG  : alarm 60 fired
10-28 17:04:30 Dummy-121  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.47,684949:4)
10-28 17:04:30 Dummy-121  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,684949:4)
10-28 17:04:30 MainThread artisanlib.main      DEBUG  : message: Event # 52 recorded at BT = 185.5CC Time = 04:00
10-28 17:04:30 MainThread artisanlib.main      DEBUG  : message: Alarm 60 triggered
10-28 17:04:30 MainThread artisanlib.main      DEBUG  : message: Event # 53 recorded at BT = 185.5CC Time = 04:00
10-28 17:04:30 Dummy-845  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,5.8999999999999995,687340:4)
10-28 17:04:30 Dummy-845  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:4)
10-28 17:04:33 MainThread artisanlib.main      DEBUG  : message: Event # 54 recorded at BT = 186.0CC Time = 04:02
10-28 17:04:33 Dummy-670  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,7.249999999999999,687340:3)
10-28 17:04:33 Dummy-670  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:3)
10-28 17:04:35 Dummy-700  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,4.55,687340:4)
10-28 17:04:35 Dummy-700  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:4)
[... cut ...]
10-28 17:34:43 MainThread artisanlib.canvas    INFO   : phidgetManager stopped
10-28 17:34:43 MainThread artisanlib.canvas    INFO   : phidgetLog stopped
10-28 17:34:43 MainThread artisanlib.canvas    INFO   : phidgetLog started
10-28 17:34:43 MainThread artisanlib.canvas    INFO   : phidgetManager started
10-28 17:34:59 MainThread artisanlib.main      INFO   : export settings to C:/Users/Redber/Documents/Artisan Roast Files/Artisan Settings.aset
10-28 17:34:59 MainThread artisanlib.main      INFO   : wrote 928 settings in 0.02s

MAKOMO, 2024-10-28:
I already base64 decoded your previous attachment and found this one

```
10-28 17:22:49 Dummy-140  artisanlib.comm      DEBUG  : phidgetVOUTsetVOUT(0,91.76,687340:3)
10-28 17:22:49 Dummy-140  artisanlib.comm      DEBUG  : phidgetVOUTattach(0,687340:3)
10-28 17:22:49 Dummy-140  artisanlib.comm      ERROR  : PhidgetException 0x15 (Invalid Argument)
Value must be in range: -10.000000 - 10.000000.
Traceback (most recent call last):
  File "artisanlib\comm.py", line 4581, in phidgetVOUTsetVOUT
  File "Phidget22\Devices\VoltageOutput.py", line 115, in setVoltage
Phidget22.PhidgetException.PhidgetException: PhidgetException 0x15 (Invalid Argument)
Value must be in range: -10.000000 - 10.000000.
````

You instruct here the VOUT phidget to set the output voltage to 91.76V. However, that Phidget only supports -10V to 10V. Couldn't find another issue in the log file. Could this be related to your problem? So maybe you should use `10-{}` in your  second slider command.

Anyhow, the issue I found with the PWM Phidget modules seems not to be related to your issue.

Robjones6146, 2024-10-28:
Thank you so much. That seems to be the issue. First one working, second
voltage too high.

Will give it a try in the morning.

closed 2024-10-28 by MAKOMO