artisan-roaster-scope/artisan #1737: Latest Version 3.0.X fails to create TIMEX in .alog (and won't display graphs / projected temps correctly) - Websocket

josarsepi, 2024-11-05:
### Problem Description

Latest Version 3.0.X fails to create TIMEX in .alog (and won't display graphs / projected temps correctly) - Websocket
- I've tried 3.0.0 & 3.0.2

I'm attaching an .alog from 2.10.4 and an .alog from 3.0.2 and an artisan-settings file.  

[artisan-settings.txt](https://github.com/user-attachments/files/17639457/artisan-settings.txt)
[Australia 2024, Mountain Top Estate Microlot_24-10-27_1549.alog.txt](https://github.com/user-attachments/files/17639458/Australia.2024.Mountain.Top.Estate.Microlot_24-10-27_1549.alog.txt)
[BrokenALog-24-10-27_1634.alog.txt](https://github.com/user-attachments/files/17639459/BrokenALog-24-10-27_1634.alog.txt)

The problem is present on both Windows 11 and Ubuntu OS but the temperature / pressure readings are present and do change over time, but time remains at 0.

### Steps To Reproduce

In Artisan 3.0.X, attempt roasting with websocket.

### Artisan Version

3.0.2

### Computer OS and Version

Ubuntu 24.04, Windows 11

### Connected devices or roasting machine

Phidget Sensors via Websocket

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-11-06:
Thanks for your report!

You are obviously running your own WebSocket server script. Any chance to share that with us to enable us to duplicate your findings?

What would also help is, if you could share your artisan.log file with us. You can find this in the following locations

```
C:\Users\<USER>\AppData\Local\artisan-scope\Artisan (Windows
C:\Users\<USER>\AppData\Local\artisan-scope\Artisan (Windows
```

MAKOMO, 2024-11-06:
Should be fixed by Commit cd8ffc9. Please confirm by testing the current [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous)

josarsepi, 2024-11-06:
Sorry for the delay!  I'll attach the websocket code I'm using here and then I'll test the continuous build and get back to you later this evening.

[phidgetWebsocketServer.py.txt](https://github.com/user-attachments/files/17653740/phidgetWebsocketServer.py.txt)
[phidgetWorker.py.txt](https://github.com/user-attachments/files/17653741/phidgetWorker.py.txt)

josarsepi, 2024-11-06:
@MAKOMO, validated that continuous build is working as expected on Windows & Linux!  Thank you for the time effort!  Artisan & artisan.plus are a great combo, and coffee wouldn't be the same without your hard work.

closed 2024-11-06 by josarsepi

MAKOMO, 2024-11-07:
Thanks for your kind words! Keep up the flame!