artisan-roaster-scope/artisan #1745: Modbuss Disconnection

BeanMonkey, 2024-11-18:
### Problem Description

Afternoon, 

We have been having this issue for a while, since version 2.8.4 where our Coffed sr15 and sr3 disconnects from the modbus after every roast. We have to restart the usb connection each time, usually by plugging and unplugging it to restart the modbus connection.
When updating to the latest artisan release it would not connect at all this morning. I have rolled back to 2.8.4 and it mostly works now.

Any ideas appreciated. 

Warmest
 

### Artisan Version

Other - include in Description

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Coffed SR15 and SR3

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-11-23:
Artisan v3.1 just got released could you please try if that resolves your issue and report back here?

Michu108, 2024-12-02:
Hello. If I can join? We have the same problem with Coffed sr5 manual. Especially after upgrading to the newest version 3.1.0 everything started to go wrong (before we used 2.4.4 for few years and everything was great). 3.1.0 worked first day only. Next day didn't want to connect at all (Modbus connection error. After we went back to the version 2.4.4 it started to work a little bit. I mean we could roast but it disconnects after every roast and an error appears when we want to use background profile. Picture attached
[image]

MAKOMO, 2024-12-02:
The Artisan team is not able to support older versions. Please upgrade to the current version which is v3.1 which also worked fine for you, at least on the first day. The question is what happend between that day and the next. The software itself has no idea on which day it is started thus your issues must be caused by some form of external (to Artisan) effect.

Please do the following. Attach the profile you just tried to load in the background from your screenshot here on GitHub to a comment field by drag-and-drop. However, before that you need to rename it from .alog to end in .txt.

Next open the file browser on your PC and point it to

```
C:\Users\<USER>\AppData\Local\artisan-scope\Artisan
```

(replace <USER> by your Windows account name). You should find a file named `artisan.log`. Rename this into `artisan.txt` and also upload it to GitHub by drag-and-drop to a comment box in this thread.

THANKS!

Michu108, 2024-12-02:
OO. Thank you. Nice trick. I will try it in a minute :)

Michu108, 2024-12-03:
I'm sorry but cannot find such a file artisan.log and I don't have such a directory 
(C:\Users\<USER>\AppData\Local\artisan-scope\Artisan) in my my Windows. 
I have a windows 10 for now. Maybe I will try to move to Windows 11 first. 

This is printscreen from the error and the profile with txt extantion
[image]

Michu108, 2024-12-03:
[image]
This is the error log from Artisan

Michu108, 2024-12-03:
[profil Brazylia.txt](https://github.com/user-attachments/files/17998867/profil.Brazylia.txt)
ANd the proflle that I wanted to open as .txt. 
Sorry for late reply but I'm not a computer geek and had to search how to do it first :)

Michu108, 2024-12-03:
At last I found the file artisan.log. There are even 2 of these files - log.1 and log.2

[artisan log 1.txt](https://github.com/user-attachments/files/17999156/artisan.log.1.txt)
[artisan log 2.txt](https://github.com/user-attachments/files/17999158/artisan.log.2.txt)

MAKOMO, 2024-12-04:
It seems you are reporting two different issues here. One is loading your attached profile in the background which fails and generates the error messages you show in the screenshots. This issue is caused by empty event type names which are no longer supported in newer Artisan versions. I just added a fix to deal better with such profiles written by older Artisan versions. A [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) containing this fix is just being build.

The MODBUS connection issues you report in the text seems not to be caused by an issue in Artisan. The log files you attached contain only data from 11-27 on which day Artisan could not establish a connection with your machine simply because the serial port indicated by your setup could not be opened. This could indicate a wrong configuration (maybe the com port the machine is connected to is not COM3), a serial driver problem (is the corresponding driver installed on that machine?), or a hardware issue (is the machine correctly connected to the serial to USB hardware and is that converter working correctly?). Nothing can be done on the Artisan side on that one. Maybe a more current log contains more information (you only attached the older log files ending with a number, but not the current one which is just named `artisan.log`).

BeanMonkey, 2024-12-04:
Apologies I have been out the loop launching all this Christmas products. 

My issue is similar with the dropped connections but I do not get any errors. The modbus does not pick up a communication between software and roaster. 

I have double checked the company port settings and baud rates for both coffed roasters we have with putty and the serial connections do open. Not sure if that's useful. 

When I upgrade to any of the 3.0/3.1 artisans unfortunately the software recognises neither roasters and I just get a modbus connection error straight away. 

Happy to provide anything needed tomorrow (roast day today) to help diagnose.

MAKOMO, 2024-12-04:
Hm. Strange. MODBUS serial communication on Windows with Artisan v3 works well for many others. There is no magic in opening that serial port and that code (nor the underlying lib) did not change for the past 7 years. What did change is the MODBUS lib itself, but that works for others as well and mostly contains fixes and speed improvements.

Could you please verify you have set the correct serial port (seems not to be COM3) in the MODBUS tab (NOT the port setting under ET/BT!!), menu Config >> Port, 3rd tab.

What could help is an artisan.log file which shows the app start up. Yours just show the port open failure over and over again as you kept Artisan ON retrying every second. Maybe you can retry and press OFF and app quit after some seconds you received the MODBUS error and then send the artisan.log file before restarting Artisan. Then the log file should contain some more information.

MAKOMO, 2024-12-04:
Yes. There are a number of errors, but some point to a permission issue.

```

11-29 13:57:45 Dummy-2    pymodbus.logging     ERROR  : Cannot configure port, something went wrong. Original message: PermissionError(13, 'Urządzenie dołączone do komputera nie działa.', None, 31)
```

I am not a Windows expert, but it seems you need somehow to allow the Artisan app to access the serial communication ports in Windows (assuming it is the correct serial port and the driver is correctly installed and generates that serial port on connecting the serial2USB converter). However, this message could also simply mean that the serial port is already in use by another process. A restart of your Windows could resolve this.

Michu108, 2024-12-04:
Ok. Thank you. I will try to resolve this somehow. 

Today is even worse. After every batch when I push OFF button it resets and I have to write down the roasting properties from the beginning. And also the function Config/Events doesn't work - I cannot push OK after I open this window, only Cancel. 
[image]

MAKOMO, 2024-12-04:
> I cannot push OK after I open this window, only Cancel.

Event type names 2 and 3 are empty. Put something in and you can press OK.

Michu108, 2024-12-06:
Hello again. I have installed Windows 11. I have reinstall Artisan for 3.1.1. I loaded Coffed sr5 manual. 
[image]
And.. its the same for now :) Failed to connect

Michu108, 2024-12-06:
WOw. Something is working. I reinstalled ch340 driver. Unistalled and installed again. And is working now. At least for now :)  There is connection and I can load background without any problems. It seems it was something wrong with the drivers.

MAKOMO, 2024-12-06:
Did you install the serial driver for your machine (see https://artisan-scope.org/machines/coffed/)? According to [this page](https://artisan-scope.org/modbus_serial/) you need to install the CH34x driver. However, your machine might be different and might require the installation of the CP210x or PL2303. Why not install of those drivers? Windows will load the one that is required once you plug-in. 

Then you need to select the correct serial port. To find out open the MODBUS tab (menu Config >> Port, 3rd tab MODBUS  and check first the items shown in the Comm Port popup WHILE THE MACHINE IS NOT CONNECTED VIA USB, then CONNECT THE MACHINE, TURN IT ON and open that Comm Port popup again. There should be one additional entry. That one is the one for WIndows and that serial driver mounted for your machine.

MAKOMO, 2024-12-06:
Oh, I was typing too slow....

closed 2024-12-06 by MAKOMO

BeanMonkey, 2024-12-06:
Evening, 

With reference to my machine. I have opened the original thread and still have a different problem to Makomo. 

Reinstallation of the CH drivers makes no difference. Installing 3.1 stops the serial communication with our SR15 completely now. I have to roll back to 2.8 to get it even half way working. 

I have attached a shot of the serial log, the .log files and can provide anything else you can think might be needed. 
This issue applies to both our SR15 and our SR3 the 15 seems to communicate at a baud rate of 115200 and the 3 at 9600 with slightly different slave settings. 
/
find logs etc below
[artisanlog.txt](https://github.com/user-attachments/files/18039904/artisanlog.txt)
[artisanlog1.txt](https://github.com/user-attachments/files/18039906/artisanlog1.txt)
[artisanlog2.txt](https://github.com/user-attachments/files/18039912/artisanlog2.txt)
[image]
[image]

BeanMonkey, 2024-12-06:
Cheers for that. I'll give it a go first thing on Monday. 
In 10 years it's my first problem with artisan and it would be just the thing that it was actually a windows issue 🫡.

Michu108, 2024-12-06:
If I can help? It may be the Windows issue. On 27.11 I installed Artisan 3.1 and it was working. In the evening that day I allowed Windows to make many new updates and next day in work - Artisan stopped connecting to Modbus and the problems started. So maybe Windows upgraded the drivers by itself. 
I have very old version of CH340 driver (v3.4) and I can send you. Just don't know how. You can just give me email and I can send you that.

Michu108, 2024-12-06:
To mr Makomo - Thank you very much for all your time and help. I really apreciate this. Will donate your program, software.