artisan-roaster-scope/artisan #1754: Incorrect location of the chart

NM12262, 2024-11-30:
### Problem Description

The background chart does not move to the left, even through the keyboard control section in the background profile window!
And when I use the cursor buttons, the "Button"s are selected and the background chart does not move.
The program also shows the errors that are in the attachment.

### Steps To Reproduce

there is no problem in this section.
[image]
[image]
[image]
[image]

### Artisan Version

3.1.0

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Unknown

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-11-30:
Could you please drag-and-drop your foreground and background profiles as .alog files (renamed first to end in .txt) to a comment field hier on the GitHub issue page such we can investigate your issue?

Thanks!

NM12262, 2024-12-01:
[Brazil Rio b2 030910 20000g AUC534 B771.alog.txt](https://github.com/user-attachments/files/17968343/Brazil.Rio.b2.030910.20000g.AUC534.B771.alog.txt)
[Brazil Rio b3 030823 20000g AUC519 B760.alog.txt](https://github.com/user-attachments/files/17968344/Brazil.Rio.b3.030823.20000g.AUC519.B760.alog.txt)

MAKOMO, 2024-12-02:
That background profile B760 is broken as its data is inconsistent. Do you remember how you produced this? Did you do anything specific with it?

As the data in B760 is broken it is displayed wrongly (shifted) and produced various issues and exceptions with many features while loaded.

A note on moving the background profile with the cursor keys, independent of the problem with B760: the cursor keys are also used to move the selected sliders as well as the selected main button. The action triggered thus depends on which widget has the keyboard input focus. To get the focus on the graph, just temporarily hide the slider section (under menu View or simply press key s to toggle).

MAKOMO, 2025-01-08:
Stale

closed 2025-01-08 by MAKOMO