artisan-roaster-scope/artisan #1756: Error upon opening Scheduler tool

SustenanceLLC, 2024-12-04:
### Problem Description

Error upon opening Scheduler.
[image]

No module named 'babel.numbers' (line:23427)

version = 3.0.0
1 09:16:03.722 Error: File "artisanlib\main.py", line 23427, in schedule File "plus\schedule.py", line 1741, in __init__ File "plus\schedule.py", line 3042, in updateScheduleWindow File "plus\schedule.py", line 2784, in updateRoastedItems File "plus\schedule.py", line 866, in __init__ File "plus\schedule.py", line 779, in __init__ File "plus\schedule.py", line 925, in getRight File "babel\dates.py", line 683, in format_date File "babel\dates.py", line 352, in get_date_format File "babel\core.py", line 905, in date_formats File "babel\core.py", line 439, in _data File "babel\localedata.py", line 137, in load File "babel\localedata.py", line 143, in load

### Steps To Reproduce

Open: View --> Scheduler

### Artisan Version

3.0.0

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Mill City MC10

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-12-04:
This is a build error in some v3.0 builds. Please download and install the current Artisan version which is 3.1. Thanks!

SustenanceLLC, 2024-12-04:
Thank you, that's what I was worried about. Both of our Mill City roasters, MC60 and MC10, drop MODBUS connection intermittently throughout the roast when we installed 3.1. Our fix was to revert to 3.0, retaining Scheduling capability.

MAKOMO, 2024-12-04:
The scheduler build issue on Windows was fixed in Artisan v3.0.2.

Best to fix Artisan 3.1 MODBUS. Could you share your log file. For this hold the ALT key on Windows (OPTION key on macOS) and click  the plus icon. This should open an email with the log files attached. We will take a look.

closed 2024-12-09 by SustenanceLLC

MAKOMO, 2024-12-09:
There is one remaining issue which does not harm much but is logged regularly (most likely at START).

```
12-09 11:08:25 Dummy-10   artisanlib.main      ERROR  : [WinError 2] The system cannot find the file specified
Traceback (most recent call last):
  File "artisanlib\main.py", line 10407, in call_prog_with_args
  File "subprocess.py", line 1026, in __init__
  File "subprocess.py", line 1538, in _execute_child
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

Most likely you have an action attached (to START?) in the Event tab which tries to run a batch script which cannot be found. Either remove this or fix the path to that script to avoid having that error logged.