artisan-roaster-scope/artisan #1763: The Korean font in the title is broken.

jcornil, 2024-12-12:
### Problem Description

The Korean font only in the title is broken.

[image]

[Default_Settings.zip](https://github.com/user-attachments/files/18106246/Default_Settings.zip)

### Steps To Reproduce

1. Open Roast > Properties
2. Set the title in Korean (ex: 한글 )
3. Click OK to save
4. Check the title at the main screen

### Artisan Version

3.1.0

### Computer OS and Version

Windows 11

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-12-12:
Thanks for your report. The font used in dialogs is some OS system default while the font used for chart title and annotations might be a different one (again depending on your OS if set to DEFAULT). On macOS the default chart font renders Korean characters correctly, on Windows this seems not to be the case.

[image]

Anyhow, you can set the chart font explicitly to one containing the Korean characters (menu Config >> Curves, last tab UI).

[image]

jcornil, 2024-12-12:
> Thanks for your report. The font used in dialogs is some OS system default while the font used for chart title and annotations might be a different one (again depending on your OS if set to DEFAULT). On macOS the default chart font renders Korean characters correctly, on Windows this seems not to be the case.
> 
> <img alt="Screenshot 2024-12-12 at 08 11 37" width="557" src="https://private-user-images.githubusercontent.com/13909011/395048116-29c2a822-e7ff-42da-8aff-e016319bb446.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5ODg3NDUsIm5iZiI6MTczMzk4ODQ0NSwicGF0aCI6Ii8xMzkwOTAxMS8zOTUwNDgxMTYtMjljMmE4MjItZTdmZi00MmRhLThhZmYtZTAxNjMxOWJiNDQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEyVDA3MjcyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzNjk5NTI4M2QyZjY1Y2Y2NjM5MzYwZTAzYTU5MTM4YWI5OWMzMTE1YTU5YjU0NGZjZDBiNGNjZjk5NjhmMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0._Pm2KBZQgeRL0IgV9wbe4Ivzn7-ZhRRcxRPU-iGwPDY">
> Anyhow, you can set the chart font explicitly to one containing the Korean characters (menu Config >> Curves, last tab UI).
> 
> <img alt="Screenshot 2024-12-12 at 08 12 18" width="646" src="https://private-user-images.githubusercontent.com/13909011/395048147-e8b9949e-2fce-48de-9c8a-05cfda9e7eca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5ODg3NDUsIm5iZiI6MTczMzk4ODQ0NSwicGF0aCI6Ii8xMzkwOTAxMS8zOTUwNDgxNDctZThiOTk0OWUtMmZjZS00OGRlLTljOGEtMDVjZmRhOWU3ZWNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEyVDA3MjcyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1OTcyNmFjYzYzMjFkMjRlNzMxYjJkNDQxMDQ0OGU5Y2MwMmI3MmE0MTdhZWUyMzk1NGFhODdhMGMzY2FiMWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.T17m5igCrEscWiia-j6-LUb7oatxtfCDQ6lUAn91q54">

Thanks, it works!

closed 2024-12-12 by jcornil