artisan-roaster-scope/artisan #1765: quirk in artisan-linux-3.1.0_raspbian-bookworm.deb

ju-emb, 2024-12-17:
### Problem Description

The official deb package for raspbian has a mismatch in architectures.
The deb package contains arm64 binaries but doesn't install on arm64.

### Steps To Reproduce

```
root@artisan:~# uname -m
aarch64

root@artisan:~# dpkg -i artisan-linux-3.1.0_raspbian-bookworm.deb
dpkg: error processing archive artisan-linux-3.1.0_raspbian-bookworm.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 artisan-linux-3.1.0_raspbian-bookworm.deb
```

it installs fine on armhf platform

```
root@debian:~# uname -m
armv7l

root@debian:~# dpkg -i artisan-linux-3.1.0_raspbian-bookworm.deb
(Reading database ... 35218 files and directories currently installed.)
Preparing to unpack artisan-linux-3.1.0_raspbian-bookworm.deb ...
Unpacking artisan (3.1.0) over (3.1.0) ...
Setting up artisan (3.1.0) ...
Processing triggers for man-db (2.11.2-2) ...
```

install .deb on armv7l  or extract the contained binary from the package and run an check, it returns aarch64

```
root@debian:~# file /usr/share/artisan/artisan
/usr/share/artisan/artisan: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, in
terpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=165c3840e46a056d08c976cddc9073109cf26ee7, for GNU/Linux 
3.7.0, stripped
```

### Artisan Version

3.1.0

### Computer OS and Version

linux

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

wvengen, 2024-12-19:
Hi, thanks for your bug report, that seems like a bug!

The architecture is set on [this line](https://github.com/artisan-roaster-scope/artisan/blob/d9e440e5dded6082faee8cd17c5b6d168df4c93b/src/build-rpi-deb.sh#L12), I think that may need to become `arm64` instead, as we are targetting 64-bit Raspbian.

wvengen, 2024-12-23:
Could you try to see if the latest [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) fixes the issue?

MAKOMO, 2024-12-23:
RPi builds are no longer on CI for reasons. Sorry. Those builds are manually build for each release. I will look into this architecture mismatch hopefully today...

wvengen, 2024-12-23:
Ah, sorry, thanks for mentioning!

MAKOMO, 2024-12-23:
I made a new deb build using 'arm64' as architecture and updated it to the 3.1 release. I updated the architecture before the release, but edited a file (now removed from the repro) which is no longer in use by the build scripts. Sorry!

Still this is a bit confusing to me as I get the following on my RPi

```
# uname -m
aarch64
# dpkg --print-architecture
arm64
```

Please re-download the build (now renamed to reflect the proper OS name) and report if this now installs and runs fine.

Thanks for your report!
M

ju-emb, 2024-12-24:
OK thanks Marko, I'll give it a try

OT
@MAKOMO 
this aarch64/arm64 stuff is just a name thing, technically is the same thing.
When the first arm 64-bit devices came out manufacturers "created" one and the other name in order to distinguish tool-chain identification, and now we have to deal with that.
General consensus is to name it aarch64, so arm64 is legacy.

Important is the difference between armhf (arm 32-bit) and aarch64/arm64 (arm 64-bit), those use different tool-chains.

So, don't be confused if one tool returns aarch64 and another arm64, it's the same architecture, but slightly older/newer tools that report the same thing.

MAKOMO, 2024-12-24:
@ju-emb : so for the next builds I will rename this architecture tag from arm64 to aarch64 to fit this consensus. Thanks for this explanation!

MAKOMO, 2024-12-24:
Ah, ok. So let's keep things as they are now. Thanks for this further clarification.

MAKOMO, 2025-01-08:
Seems to be resolved

closed 2025-01-08 by MAKOMO