artisan-roaster-scope/artisan #1781: Two roast properties windows, changes not saved

lookingafar, 2025-01-17:
### Problem Description

See "Steps to reproduce"

### Steps To Reproduce

1. Check "Open on drop" in roast properties, click "Ok" to close the window.
2. Start a roast
3. Open roast properties window
4. Let autodrop trigger (by dropping the beans or lowering the temperature enough).
5. The second roast properties window opens.
6. Now you have two roast properties windows, one on top of another (which shouldn't happen, I believe).
7. Modify roast properties in the last opened window and click "Save". The window closes, but another roast properties window is still there.
8. Click "Cancel" to close the window.
9. Open another roast properties window. The changes you made earlier have not been saved.

### Artisan Version

3.1.0

### Computer OS and Version

Linux Manjaro (5.15.164-1)

### Connected devices or roasting machine

HB M5

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [x] No files are attached.

MAKOMO, 2025-01-17:
Thanks for your detailed report! Should be fixed now. The fix will appear in the next development build and will be part of the next official release.

closed 2025-01-17 by MAKOMO

lookingafar, 2025-01-23:
Thanks, Marko.