artisan-roaster-scope/artisan #1791: Hot Top disconnecting since 3.1.0 update.

gevans3, 2025-02-01:
### Problem Description

I just upgraded to Artisan 3.1.0 for MAC (Sequoia 15.2) driving my HotTop +2 and i have been having sporadic issues with the Hot Top disconnecting from artisan  in the middle of a roast.  I had no issues prior to the update and I do have filters on my USB to mitigate any electrical interference.  I also run Artisan with my laptop on battery, to eliminate plug in power being a source of any interference.  One roast will go flawlessly, the next will simply stop, creating some challenges to say the least in salvaging the roast (an expensive proposition given that I was roasting a Lamastus Elida Estates Gesha when it shit the bed this morning).  Furthermore, reconnecting was a challenge and I ultimately had to clear everything and reconfigure the machine from the ground up to get it to reconnect and work normally.  I think I was able to salvage the roast as I did an emergency dump and effectively picked up the roast where I left off (more or less) and ran the whole thing with another profile as a background reference.  Apologies, I did not think to save the truncated roast log.

### Steps To Reproduce

1. Roast terminated at about the 3:30 mark
2. Emergency bean dump
3. Attempted to reconnect but had to reinitialize the roaster as well as close out and relaunch Artisan and configure to HotTop and then reconnect.  In essence, I had to reset everything and start from square one.
4. This has only been happening since the Artisan 3.1.0 update on MAC Sequoia 15.2

### Artisan Version

3.1.0

### Computer OS and Version

MAC OS 15.2

### Connected devices or roasting machine

HotTop +2 model

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [x] No files are attached.

MAKOMO, 2025-02-01:
Strange. No issues here on the same setup. Nothing changed regarding the Hottop communication the last few Artisan releases and Artisan v3.1 is going strong for everybody. No real issues reported since its release.

gevans3, 2025-02-01:
Thanks for looking into it.  Maybe I got some kind of power fluctuation…I
live in Panama and that can happen on the rare occasion.  Just a
coincidence that today was my first time roasting with the 3.1.0 update…the
first time I have had any connection issues in more than a year.

I am in the middle of trying to translate roast profiles from my Link
roaster to my Hot Top, so it was mildly frustrating.  It could have at
least waited until I was roasting a typica for one of my blends…lmao.

Thank you again.  Loving my Hot Top.

Geo

closed 2025-02-03 by MAKOMO

gevans3, 2025-02-03:
> Everything is possible, however, it is unlikely that this disruption was
> cased by a bug in Artisan, as many people are using your combination
> without issues. Thus I close this for now. If you get additional hints on
> the cause, best something reproducable, please reopen the issue and we will
> dig this down.

Roger that! Thank you!

Geo