artisan-roaster-scope/artisan #1805: Bluetooth dialog does not work with roaster on macOS

neuralldev, 2025-02-28:
### Problem Description

On MacOS Sequoia V3.3.1 15.3.1 (24D70) with M2 processor
connecting to Itop Cyberroaster (V2)

Start Artisan, press "ON"
Bluetooth is established on the Mac, Artisan 3.1.0 detects /dev/cu.AKROASTER virtual device.
when running dialog, serial log displays -1 all the time:

`
4 08:06:12.847 ArduinoTC4: /dev/cu.AKROASTER,115200,8,N,1,0.6 || Tx = READ || Rx = ['-1'] || Ts= -1.00, -1.00, -1.00, -1.00, 0.00, 0.00

3 08:06:12.846 ArduinoTC4: Tx = READ || Rx = 

2 08:06:02.123 ArduinoTC4: /dev/cu.AKROASTER,115200,8,N,1,0.6 || Tx = CHAN;2100 || Rx = 

1 08:06:00.918 serial port opened: /dev/cu.AKROASTER,115200,8,N,1,0.6
`

I tried to run V2.10.4 and got the same result.

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-02-28:
The Cyberroaster is not supported by Artisan. You seem trying to connect it using the serial TC4 protocol via a serial connection over Classic Bluetooth and Artisan is not receiving any (or not the expected) response from the machine.

closed 2025-02-28 by MAKOMO

Terracotta-6, 2025-02-28:
Skywalker Delta coffee roaster are not on the list supported by artisan, so it's best to contact the device manufacturer.