artisan-roaster-scope/artisan #1811: Santoker RX300 (same system R Master Series ) Temperature is not read

Tawan-Tan, 2025-03-13:
### Problem Description

I connected my Santoker RX300 to Artisan by setting Config >> Machine >> Santoker >> R Master Series. It shows that it is connected, but the temperature on the right side cannot be read, and I cannot control the roaster, such as adjusting the burner or damper.

[image]

### Artisan Version

3.1.0

### Computer OS and Version

macOS Sequoia 15.3.2

### Connected devices or roasting machine

Santoker RX300 the same system as R Master Series

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-03-14:
I made several changes which might resolve this and added some extra logging to help investigating this. Could you please check if this build

https://github.com/artisan-roaster-scope/artisan/releases/tag/santoker_test

connects and displays ET/BT temperatures?

If not, please activate debug logging by holding the OPTION + COMMAND keys while clicking on the plus icon in the upper left corner of the Artisan main window and tick device "Logging" under "Config >> Devices". Then press ON and wait one or two minutes. Finally, click again the plus icon, but hold only the OPTION key. This should open an email window with the log files attached. Send those log files to `logfile@artisan.plus` (or rename them to end in .txt and drag-and-drop them here on GitHub on a comment field.

Tawan-Tan, 2025-03-14:
Thank you for solving the problem. The temperature is read, but as can be read in the log file, we still cannot control the burner, air, and drum.

[artisan.txt](https://github.com/user-attachments/files/19253064/artisan.txt)
[artisan_device.txt](https://github.com/user-attachments/files/19253065/artisan_device.txt)

MAKOMO, 2025-03-16:
Thanks for testing and your detailed review. I updated the build under the same link. Could you please test this one too and report back again? Thanks a lot!

MAKOMO, 2025-03-25:
Hm, this seems to be stall. No success confirmation. I will close this one.

closed 2025-03-25 by MAKOMO

Tawan-Tan, 2025-03-25:
Sorry for the late reply, after testing for a week, everything went well.