artisan-roaster-scope/artisan #1817: ROR curve smoothing problems

RickGroszkiewicz, 2025-04-02:
### Problem Description

I recently upgraded to artisan 3.1.0, and noticed a BIG difference. On my Hottop B-2K+, the ROR curve is noticeably less smooth than under artisan 2.8.4. I made several changes to Curves / Filters, but it still looks much worse to me. 

These screen shots show the changes
https://softwarepolish.com/albums/Artisan%20software%202.8%20-%20Linux%20Mint%2021%20-%20%20Hottop%20B-2K/#img=Screenshot%20from%202023-11-05%2017-14-49.png

https://softwarepolish.com/albums/Artisan%20software%203.1%20-%20Linux%20Mint%2021%20-%20%20Hottop%20B-2K/index.html#img=B2K29.png

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

RickGroszkiewicz, 2025-04-02:
Here are the screenshots of the two roasts

[image]

[image]

MAKOMO, 2025-04-02:
I see the noise already on your ET and BT signals!

There have not been any changes on the RoR calculation or filtering nor the Hottop communication between those versions. I see considerable noise on both roasts if all smoothing is turned off. 

Note that the Hottop delivers all data without any decimals which leads to digital noise. The only counter measure you have is to enable "Interpolate Duplicates" under Input Filter of the Curves dialog which will substitute the second of two succeeding readings of exactly the same value by an interpolated value computed from the first of those and the next reading. 

I checked the last roast done on my Hottop 2k+ (from 2014) which has smoother readings than yours.

I close this as this seems not to be a software issue.

closed 2025-04-02 by MAKOMO

beanoccio, 2025-05-23:
The problem is: Nobody knows exactly what all these settings in the "Curves" do. All we can do is guess and try.

RickGroszkiewicz, 2025-05-23:
Challenge accepted! I will try to put some notes to paper.

roasterdave, 2025-05-23:
Excellent!

roasterdave, 2025-05-24:
@beanoccio Sorry, this is an out of scope question in a closed issue.  Please start a new [discussion](https://github.com/artisan-roaster-scope/artisan/discussions/new?category=q-a) and we can discuss it there.  Be sure to post your settings there (Help>> Save Settings, then add ".txt" to the file name and drag the file onto the message box on the Discussion page).  Also post a sample alog file (add ".txt" to that file name as well).