artisan-roaster-scope/artisan #1839: slider event is not working on 3.1.2

mxofelix, 2025-05-01:
### Problem Description

Hi Marko,
i had just updated my artisan to from 3.1.0 to 3.1.2.
i had run into issue that slider event is not working anymore.
i am connecting via MODBUS.

but the button event still works.
therefor i cannot use the ramp/soak function for now.

[image]

[artisan-settings 010525.aset.txt](https://github.com/user-attachments/files/19987094/artisan-settings.010525.aset.txt)

Regards,
Felix

### Steps To Reproduce

1. ON
2. move the adjustment from manual to auto on my machine
3. click on slider button

### Artisan Version

Other - include in Description

### Computer OS and Version

WIndows 11

### Connected devices or roasting machine

Berto essential 1kg

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [x] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-05-01:
Replace the generic MODBUS `write` command (which writes a float value to two 16bit registers) by the specific `writeSingle` command (which writes any value rounded to one 16bit register) in your sliders definition.

[image]

mxofelix, 2025-05-01:
thank you!
it works now..

MAKOMO, 2025-05-01:
Thanks for the fast test and confirmation!

closed 2025-05-01 by MAKOMO