artisan-roaster-scope/artisan #1843: Missing releases breaking package manager

Bohreromir, 2025-05-02:
**Is your feature request related to a problem? Please describe.**
I recently added artisan to the nix package manager. Unfortunately old releases are missing in this repo and now we cant depend on them in our builds.

**Describe the solution you'd like**
I fully understand your stance on not providing support for older releases, but ask you to not delete them from github anyway.
Some users depend on having a stable os with stable packages. If it where to happen that an artisan release is broken user currently do not have a way to continue working with a working version.
As an alternative you might want to add a warning about only supporting the latest version to older releases.

**Describe alternatives you've considered**
I'd rather not build artisan from source for every release. Alternatives might be trying to get old releases of artisan from archive.org.

**Additional context**
initial commit: https://github.com/NixOS/nixpkgs/pull/393753
package update that broke shortly after because of missing releases: https://github.com/NixOS/nixpkgs/pull/403071
discussion about the broken builds: https://github.com/NixOS/nixpkgs/pull/403636

mrpenner, 2025-05-02:
Is that common practice to repackage binaries for Nix? The Linux distros I'm familiar with are quite strong that they build from source themselves.

Bohreromir, 2025-05-05:
At least not too uncommon for appimages.
I'm working on building it from source now, so that wont break nix again.
But even as a "regular" user I'd expect to be able to access old releases here.

Bohreromir, 2025-05-09:
As a user I'd like to be able to easily roll back in the case of breaking changes or bugs.
Feel free to close this issue as I have set up the build for nix.

closed 2025-05-15 by MAKOMO