artisan-roaster-scope/artisan #1845: Checksum error on Artisan 3.1.4 for homebrew on Mac OSX

MrJJH1, 2025-05-06:
### Problem Description

When upgrading artisan to 3.1.4 I get a checksum error. 
I have tried upgrading 2 times and also removing all artisan and run the install - brew install --cask command.

Following error happens
Error: SHA256 mismatch
Expected: 5c6988024d201e5658970f630a048a8b6e2dab5e2f452ff0bc000dd05e36b88a
  Actual: d48f4d978a8eb7aae6d0076976b99a2ecd8a1ed8ec564e63eb01c748cab16f16
    File: /Users/User/Library/Caches/Homebrew/downloads/4c5a0c2f5200440b1698b0ca2c66999f6a06f851e0e9792202017ab2ed14ee0e--artisan-mac-3.1.4.dmg

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-05-08:
We have nothing to do with homebrew. We publish our builds here on GitHub. I have no idea who is responsible for the homebrew builds. Ask on homebrew for help.

closed 2025-05-08 by MAKOMO

MAKOMO, 2025-05-08:
Hm. You are right. I have no idea how this comes and who added this. Not me. I will investigate.