artisan-roaster-scope/artisan #1846: Updated to 3.14, no recognition from roasters

BeanMonkey, 2025-05-07:
### Problem Description

Hello all, 

I have been having this issue for a while and asked once before but never got a resolution. (someone else who jumped on the thread with a slightly different issue did though). I can run both my roasters on Artisan 2.8.4, though between roasts any usb interface must be unplugged and plugged back in for either roaster to reregister the modbus.
If I run any version above this (which I have tried with every new release) the modbus will not connect or recognise either roaster. I literally press on/start and it just pops up a not recognised error for both roasters I am at a little loss these days on what to do with this. 

Will supply screen shots of the setup files (com8 is SR15 and com3 is the SR3) and can get anything else that is needed. 

Kindest

[image]

[image]

### Artisan Version

3.1.4

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Coffed manual SR15 and Coffed delta+ SR3

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

Terracotta-6, 2025-05-07:
When replacing the USB-to-485 with the CP2102 chipset, the compatibility of the CH340 chip is not as good as that of the CP2102.

MAKOMO, 2025-05-08:
> Purchase a new USB-to-485 converter using the CP2102 or FT232RL FT2232HL chipset, The compatibility of the CH340 chip is not as good as that of the CP2102.

Yes, this is what we found about this issue as well. There seem to be some USB-to-485 converters that don't play well with many Artisan versions. All reports we received were based on the CH340 chipset connected to Windows. This might also be a CH340 driver problem. Most likely some timing problems. Newer Artisan versions integrate MODBUS libraries that are more strictly implementing the timing specified by the MODBUS protocol than earlier Artisan version. Similar issues have not reported by any converters based on the CP2102 or FTDI chipsets. There is not much we can do on the Artisan side to work around those CH340/Windows issues. Sorry.

BeanMonkey, 2025-05-08:
Afternoon all, 

I've done a little more reading based on your comments and found an someone local to talk to about it and we will look to replace the adapter interface with the FTDI ones and take it from there. Thank you for the heads up on that other chipset not being supported. That's just what the roasters shipped with a few years ago.

Warmest

MAKOMO, 2025-05-08:
It is not that the CH340 is not supported by Artisan. Support for serial drivers is a matter of the operating system. The issue seems that those cheaper modules either have some hardware flaws or their drivers are buggy (at least on some operating systems) or not well-maintained by the producer.

closed 2025-05-08 by MAKOMO