artisan-roaster-scope/artisan #1847: unhandled exception while trying to access Devices menu

neuralldev, 2025-05-08:
### Problem Description

it happened at suddent, I closed the app and repopened it after some time and I cannot access the Device menu anymore. I tried different aset files, without success. It normally operates on TC4. 

Here is error message from menu -> 
`version = 3.1.4

3 14:17:24.343 Error: File "/Applications/Artisan.app/Contents/Resources/lib/python3.13/artisanlib/main.py", line 24208, in deviceassigment File "/Applications/Artisan.app/Contents/Resources/lib/python3.13/artisanlib/devices.py", line 1492, in __init__ -------------------------------------------------------------------------------- /Applications/Artisan.app/Contents/Resources/lib/python3.13/artisanlib/main.py::deviceassigment:24208 self _False DeviceAssignmentDlg /Applications/Artisan.app/Contents/Resources/lib/python3.13/artisanlib/devices.py::__init__:1492 self parent aw activeTab0 dev['Omega HH806AU', 'Omega HH506RA', 'CENTER 309', 'CENTER 306', 'CENTER 305', 'CENTER 304', 'CENTER 303', 'CENTER 302', 'CENTER 301', 'CENTER 300', 'VOLTCRAFT K204', 'VOLTCRAFT K202', 'VOLTCRAFT 300K', 'VOLTCRAFT 302KJ', 'EXTECH 421509', 'Omega HH802U', 'Omega HH309', 'NONE', 'TE VA18B', 'Omega HHM28[6]', 'Program', 'MODBUS', 'VOLTCRAFT K201', 'Amprobe TMD-56', 'Phidget 1048 4xTC 01', 'Phidget 1046 4xRTD 01', 'Mastech MS6514', 'Phidget IO 01', 'Yocto Thermocouple', 'Yocto PT100', 'Phidget 1045 IR', 'DUMMY', 'Phidget 1051 1xTC 01', 'Hottop BT/ET', 'Apollo DT301', 'EXTECH 755', 'Phidget TMP1101 4xTC 01', 'Phidget TMP1100 1xTC', 'Phidget 1011 IO 01', 'Phidget HUB IO 01', 'VOLTCRAFT PL-125-T2', 'Phidget TMP1200 1xRTD A', 'Phidget IO Digital 01', 'Phidget 1011 IO Digital 01', 'Phidget HUB IO Digital 01', 'VOLTCRAFT PL-125-T4', 'S7', 'Aillio Bullet R1 BT/DT', 'Phidget DAQ1400 Current', 'Phidget DAQ1400 Frequency', 'Phidget DAQ1400 Digital', 'Phidget DAQ1400 Voltage', 'Aillio Bullet R1 IBTS/BT', 'Yocto IR', 'Behmor BT/CT', 'VICTOR 86B', 'Phidget HUB IO 0', 'Phidget HUB IO Digital 0', 'Yocto 4-20mA Rx', 'WebSocket', 'HB BT/ET', 'Yocto 0-10V Rx', 'Yocto milliVolt Rx', 'Yocto Serial', 'Phidget VCP1000', 'Phidget VCP1001', 'Phidget VCP1002', 'ARC BT/ET', 'Yocto Power', 'Yocto Energy', 'Yocto Voltage', 'Yocto Current', 'Yocto Sensor', 'Santoker BT/ET', 'Phidget DAQ1500', 'Kaleido BT/ET', 'IKAWA', 'Phidget DAQ1000 01', 'Phidget DAQ1200 01', 'Phidget DAQ1300 01', 'Phidget DAQ1301 01', 'Digi-Sense 20250-07', 'Extech 42570', 'Mugma BT/ET', 'Phidget TMP1202 1xRTD A', 'ColorTrack Serial', 'Santoker R BT/ET', 'ColorTrack BT', 'Thermoworks BlueDOT', 'Aillio Bullet R2'] limit176 _Aillio Bullet R2 i3 selectprogrambutton db_help help_button help_text_translatedHelp helpprogrambutton selectoutprogrambutton controllabel supported_ET_pids[('Fuji PXF', 4), ('Fuji PXG', 0), ('Fuji PXR', 1), ('Delta DTA', 2)] item cp0 btlabel supported_BT_pids[('', 2), ('Fuji PXF', 4), ('Fuji PXG', 0), ('Fuji PXR', 1), ('Delta DTA', 3)] label1 label2 unitids['1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26', '27', '28', '29', '30', '31', '32'] arduinoChannels['None', '1', '2', '3', '4'] arduinoETLabel arduinoBTLabel arduinoATLabel arduinoTemperatures['None', 'T1', 'T2', 'T3', 'T4', 'T5', 'T6'] FILTLabel spinBox labelETadvanced labelBTadvanced symbolicHelpButton db_reset db_reset_button reset_text_translatedReset resetButton extradevHelpButton phidgetProbeTypeItems['K', 'J', 'E', 'T'] phidgetBox1048 changeTriggersCombo model changeTriggerItems[, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ] width75 asyncFlag probeTypeCombo probeTypeItems[, , , ] rowLabel dataRateItems[, , , , , , , ] typeLabel asyncLabel changeTriggerLabel rateLabel phidget1048HBox phidget1048VBox phidget1048GroupBox phidgetBox1045 EmissivityLabel phidget1045VBox phidget1045GroupBox phidgetBox1046 gainCombo gainItems[, , , , , ] formulaCombo formulaItems[, , ] gainLabel formulaLabel phidget1046HBox phidget1046VBox phidget1046GroupBox phidgetBox1200 phidgetBox1200_2 wireItems[, , ] wireLabel changeLabel typeLabel2 wireLabel2 asyncLabel2 changeLabel2 rateLabel2 phidget1200HBox phidget1200VBox phidget1200HBox_2 phidget1200VBox_2 phidget1200_tabs phidget1200_tab1_widget phidget1200_tab2_widget phidgetGroupBoxLayout phidget1200GroupBox powerLabel modeLabel phidgetBox1400 phidget1400HBox phidget1400VBox phidget1400GroupBox phdget10481045GroupBoxHBox phidgetBox1018 dataRatesCombo voltageRangeCombo voltageRangeItems[, , , , , , , , , , ] ratioFlag dataRateLabel ratioLabel rangeLabel phidget1018HBox phidget1018GroupBox phidgetServerIdLabel phidgetPasswordLabel phidgetPortLabel phidgetServerBox phidgetPasswordBox phidgetPortBox phidgetNetworkGrid phidgetNetworkGroupBox phidget10451018HBox phidgetVBox yoctoServerIdLabel YoctoEmissivityLabel yoctoServerBox yoctoNetworkGrid yoctoNetworkGroupBox yoctoIRGrid yoctoIRHorizontalLayout yoctoAsyncGrid yoctoAsyncHorizontalLayout yoctoAsyncGroupBox yoctoIRGroupBox yoctoVBox temperatureDeviceLabel humidityDeviceLabel pressureDeviceLabel elevationLabel ambientGrid ambientHBox ambientVBox regexhostPyQt6.QtCore.QRegularExpression('(^(([0-9]|[1-9][0-9]|1[0-9]{2}|2[0-4][0-9]|25[0-5])\\.){3}([0-9]|[1-9][0-9]|1[0-9]{2}|2[0-4][0-9]|25[0-5])$)|(^(([a-zA-Z0-9]|[a-zA-Z0-9][a-zA-Z0-9\\-]*[a-zA-Z0-9])\\.)*([A-Za-z0-9]|[A-Za-z0-9][A-Za-z0-9\\-]*[A-Za-z0-9])$)') santokerHostLabel santokerPortLabel kaleidoHostLabel kaleidoPortLabel mugmaHostLabel mugmaPortLabel colorTrackMeanLabel colorTrackMedianLabel santokerNetworkGrid santokerSerialHBox santokerHBox santokerVBox santokerNetworkGroupBox kaleidoNetworkGrid kaleidoNetworkGroupBox kaleidoHBox kaleidoVBox mugmaNetworkGrid mugmaNetworkGroupBox mugmaHBox mugmaVBox colorTrackNetworkGrid colorTrackNetworkGroupBox colorTrackHBox colorTrackVBox PIDgrid PIDBox PIDGroupBox filtgrid filtgridBox arduinogrid arduinogridBox arduinoBox arduinoGroupBox programlayout programGroupBox adjustmentHelp adjustmentGroupBox adjustmentsLayout deviceSubSelector deviceSelector grid buttonLayout tab1Layout bLayout tab2Layout tab3Layout tab4Layout tab5Layout tab6Layout tab7VLayout tab7V2Layout tab7Layout tab8Layout tab8LayoutOFF scale1ModelLabel __class__ 
`
[image]

[artisan v2 tilau.txt](https://github.com/user-attachments/files/20102638/artisan.v2.tilau.txt)

### Steps To Reproduce

1. Start Artisan
2. Select Config/Device ...
3. crash

### Artisan Version

3.1.4

### Computer OS and Version

macos 15.4.1 (24E263)

### Connected devices or roasting machine

none

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [x] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

neuralldev, 2025-05-08:
In addition if I press RESET button, then try to go back to Devices menu, Artisan hangs and I have to kill it from Apple tasks list

MAKOMO, 2025-05-08:
This one is very strange. I cannot reproduce and at that line number no magic happend that could trigger such an issue. Could you please send me your artisan.log file? Hold the ALT button while clicking on the plus icon in the upper left corner of the Artisan window.

What happens if you do a Factory Reset and load your settings file? Does the device dialog still fails to open?

MAKOMO, 2025-05-08:
Could it be that the settings file you attached was not saved from your current Artisan version and is somehow older?

neuralldev, 2025-05-08:
the setting file was imported from previous release 3.1.2 and I saved it a lot of time since then. 
if I factory reset, the menu comes back.
if I load the aset it crashes again with same error message. so there is something that happens in my aset files, I tried 3 different versions I have stored. 

[artisan.log](https://github.com/user-attachments/files/20103600/artisan.log)

[artisan-settings.txt](https://github.com/user-attachments/files/20103618/artisan-settings.txt)

the one attached is for sure 3.1.4 I was working with it an hour before.

MAKOMO, 2025-05-08:
I can now duplicate the issue and fixed it. I will make a build, sign and notarize it for you to test. This takes a moment. Will post the link here.

MAKOMO, 2025-05-08:
Here is a link to a beta build which should fix this issue

https://www.dropbox.com/scl/fi/xsovta4agdvu2rcr5kmvp/artisan-mac-3.1.5-1.dmg?rlkey=h2f9m2juib2y388qznw4fxav6&dl=0

Please report!

neuralldev, 2025-05-08:
thank you very much for the fix makomo, it is fixing the issue,everything seems to work as planned :) 

let's close the case

closed 2025-05-08 by neuralldev

MAKOMO, 2025-05-08:
Thanks for testing and reporting!