artisan-roaster-scope/artisan #1851: Kaleido marking drop event at 6:59 into roast without autodetect - buttons greyed out

EdGoesBoom, 2025-05-12:
### Problem Description

I also experienced the same bug as in #1808 where Drop event was detected at Charge. That is resolved for me also in version 3.1.2.  Now I am getting a Drop event detected at 6:59 but I do not have auto detection of drop enabled.

[image]

[image]

### Steps To Reproduce

I'm including .alogs for Roast #2 which did have issue as well as Roast #3-4 of a day's session. 

[R2 No Issue Guatemala Libertad Aurelio del Cerro_295g_178.8_14.2_2025-05-07_1417 copy.alog.txt](https://github.com/user-attachments/files/20175854/R2.No.Issue.Guatemala.Libertad.Aurelio.del.Cerro_295g_178.8_14.2_2025-05-07_1417.copy.alog.txt)
[R3 Issue Guatemala Libertad Aurelio del Cerro_295g_179.2_13.6_2025-05-07_1435 copy.alog.txt](https://github.com/user-attachments/files/20175853/R3.Issue.Guatemala.Libertad.Aurelio.del.Cerro_295g_179.2_13.6_2025-05-07_1435.copy.alog.txt)
[R4 Issue same time as R3 Brazil Dry Process Sao Joaquim Farm_295g_178.5_12.5_2025-05-07_1452 copy.alog.txt](https://github.com/user-attachments/files/20175852/R4.Issue.same.time.as.R3.Brazil.Dry.Process.Sao.Joaquim.Farm_295g_178.5_12.5_2025-05-07_1452.copy.alog.txt)

As well as my settings file. 

[artisan-settings.aset.txt](https://github.com/user-attachments/files/20175886/artisan-settings.aset.txt)

### Artisan Version

3.1.2

### Computer OS and Version

MacOS 15.4.1

### Connected devices or roasting machine

Kaleido M2 Dual (connected via BT)

### Files

- [x] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [x] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

roasterdave, 2025-05-13:
This looks like the Kaledo is sending an inaccurate DROP event as explained [here in issue 1808](https://github.com/artisan-roaster-scope/artisan/issues/1808#issuecomment-2783313922).  The DROP mark at 6:59 makes sense given the Kaleido DROP state is ignored for the first 7 minutes in v3.1.2 and v3.1.4.  The 1 second difference has to do with latency compensation that pushes the mark back in time.  

The best suggestion for now is to click the DROP button to undo the errant event as described in the post linked above.

Kevin-cell-cloud, 2025-05-14:
Kalaido's dual-system model comes with a Tablet, which is a separate control terminal that synchronizes with the Artisan. The Tablet does not have "Auto Charge" and "Auto Drop" functions, all events are marked manually. In the above case I would like someone to help me to see if the "Drop" event button on the Tablet is automatically pressed when the "Drop" flag is triggered by mistake on the Artisan (they should actually be synchronized, by EV command). When the user is using Artisan exclusively, the Tablet just receives status messages and doesn't send any control commands, so there's no reason to broadcast any EV commands to Artisan, which instead sends EV commands to the Tablet to set the correct state of the event buttons on the Tablet to be pressed. It's also a good idea not to connect the Tablet that comes with the device, we can focus entirely on the roaster and Artisan.
Also, all the cases I've seen have background curves loaded. Let's try to reproduce this fault using the simplest way possible, without loading any background curves.

EdGoesBoom, 2025-05-27:
I just did 6 roasts all with background curves loaded and all with the tablet not plugged in.  None of them had the unexpected drop event.

Sorry that I did opposite of the requested testing but I needed to catch up on roasting.

roasterdave, 2025-05-27:
@EdGoesBoom This is quite useful, thank you.