artisan-roaster-scope/artisan #1855: sampling rate for Hottop roasters

RickGroszkiewicz, 2025-05-18:
The default sampling rate for the Hottop B-2K+ roaster was changed to 1 second recently (Config / machine / Hottop / KN-8828B-2K+). This produces noticeably less smooth results for the ROR curve.

I suggest that the default sampling rate for the Hottop B-2K+ should be 2 or 3 seconds.

I recently changed the default sampling rate for my Hottop B-2K+ and Hottop TC4C to be 4 seconds. This produced much better ROR curves.

Here are screen shots for both roasters. The background ROR curves were NOT sooth at all:

[image]

[image]

MAKOMO, 2025-05-18:
Could you please attach those two profiles as .alog files?

RickGroszkiewicz, 2025-05-18:
[East Africa Nye-Nye SWP Decaf 2024-roast1 roasted on 25-05-18.alog.txt](https://github.com/user-attachments/files/20275557/East.Africa.Nye-Nye.SWP.Decaf.2024-roast1.roasted.on.25-05-18.alog.txt)
[East Africa Nye-Nye SWP Decaf 2024-roast2 roasted on 25-05-18.alog.txt](https://github.com/user-attachments/files/20275558/East.Africa.Nye-Nye.SWP.Decaf.2024-roast2.roasted.on.25-05-18.alog.txt)

MAKOMO, 2025-05-19:
Instead of increasing the sampling rate you can as well increase the Delta Span (of course only to its max of 30sec). See the Appendix of my post [On Idle Noise](https://artisan-roasterscope.blogspot.com/2019/03/on-idle-noise.html) for a formal proof.

RickGroszkiewicz, 2025-05-19:
I understand - the background curves are not smooth due to different sampling.

I assume that most artisan users don't understand all of the settings under the Curves Filters dialog - there are TOO many options. I always copy the previous settings when I upgrade to a new version of artisan. 

I was concerned about two years ago when my ROR curves started looking worse. For comparison, here are two decaf roasts from 2019.

[image]

[image]

closed 2025-05-20 by MAKOMO