artisan-roaster-scope/artisan #1859: I have a problem to read extra devices with values < integer 18

beanoccio, 2025-05-21:
### Problem Description

I have a strange problem.
If the passed value is < 18, the LCD displays "uu."
If the value is >= 18, the value is displayed.

[image]

### Artisan Version

3.1.5 Continuous Build

### Computer OS and Version

W11

### Connected devices or roasting machine

Arduino mega / Modbus RTU

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-05-21:
Do you have an input filter set?

closed 2025-05-21 by beanoccio

MAKOMO, 2025-05-21:
The input filters apply to all raw inputs