artisan-roaster-scope/artisan #1862: auto DROP should not be ON for the Hottop

RickGroszkiewicz, 2025-05-25:
The default configuration for the Hottop (Config / machine / Hottop) turns the auto DROP setting ON (Config / Events / Config). Based on my experience, auto DROP should not be ON for the Hottop. 

Since the Hottop only returns integer data values, it seems likely that the coffee will be dropped prematurely. This happened to me three times recently when I mistakenly had set auto DROP to ON:

https://github.com/artisan-roaster-scope/artisan/discussions/1849

I suggest that the auto DROP setting should be OFF for the Hottop.

beanoccio, 2025-05-25:
Did you set it to "Sensitive" or "Standard"?

RickGroszkiewicz, 2025-05-26:
It was the default, which is Standard.

closed 2025-06-06 by MAKOMO