artisan-roaster-scope/artisan #1863: Backround load doesn't work

Svatopluk72, 2025-05-26:
### Problem Description

I can't upload a sample roasting profile using Background Load. The feature stopped working today. It was working until noon today.
I can't upload attachments either

[image]

### Steps To Reproduce

Issue description:
version = 3.1.4

4 18:53:35.050 Error: File "artisanlib\main.py", line 24668, in background File "artisanlib\background.py", line 125, in __init__ -------------------------------------------------------------------------------- artisanlib\main.py::background:24668 self _False backgroundDlg artisanlib\background.py::__init__:125 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: DT', 'B4: SV', 'B5: Drum', 'B6: Air'] i2 __class__ 

3 18:17:54.502 Error: File "artisanlib\main.py", line 24668, in background File "artisanlib\background.py", line 125, in __init__ -------------------------------------------------------------------------------- artisanlib\main.py::background:24668 self _False backgroundDlg artisanlib\background.py::__init__:125 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: DT', 'B4: SV', 'B5: Drum', 'B6: Air'] i2 __class__ 

2 18:17:09.227 Error: File "artisanlib\main.py", line 24668, in background File "artisanlib\background.py", line 125, in __init__ -------------------------------------------------------------------------------- artisanlib\main.py::background:24668 self _False backgroundDlg artisanlib\background.py::__init__:125 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: DT', 'B4: SV', 'B5: Drum', 'B6: Air'] i2 __class__ 

1 18:16:02.302 Error: File "artisanlib\main.py", line 24668, in background File "artisanlib\background.py", line 125, in __init__ -------------------------------------------------------------------------------- artisanlib\main.py::background:24668 self _False backgroundDlg artisanlib\background.py::__init__:125 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: DT', 'B4: SV', 'B5: Drum', 'B6: Air'] i2 __class__ 

### Artisan Version

3.1.4

### Computer OS and Version

Windows

### Connected devices or roasting machine

Ghibli R-15

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

Svatopluk72, 2025-05-26:
[image]

MAKOMO, 2025-05-26:
In the best case, please also upload your artisan-settings.aset file (save via menu Help >> Save Settings) and rename to end in .txt before drag-drop it here on a comment field.

THANKS!

Svatopluk72, 2025-05-26:
[Superiore 250526_253.txt](https://github.com/user-attachments/files/20444438/Superiore.250526_253.txt)

[artisan-settings 26052025.txt](https://github.com/user-attachments/files/20444439/artisan-settings.26052025.txt)

Svatopluk72, 2025-05-26:
Thank you Marko

Svatopluk72, 2025-05-27:
First of all, thank you very much for your help. I removed that device. And it works. I added it there because I was trying to set up the ability to control my Ghibli R-15 roaster using a slider in collaboration with ChatGPT. And the fan and drum. I can't do that. On the left side, the drum and fan values ​​are loaded correctly. On the right, the sliders show wrong values. But controlling my roaster using a slider is probably not possible, so it doesn't matter that much.
But I would like to ask a favor. An external thermometer Almemo 2590 to measure the temperature of the beans. Is there any way to set it in Artisan to take the temperature from this sensor?

Svatopluk72, 2025-05-27:
Well that's strange. It's reporting the same problem again. Can't load background

Svatopluk72, 2025-05-27:
version = 3.1.4

1 08:05:50.902 Error: File "artisanlib\main.py", line 24668, in background File "artisanlib\background.py", line 125, in __init__ -------------------------------------------------------------------------------- artisanlib\main.py::background:24668 self _False backgroundDlg artisanlib\background.py::__init__:125 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: DT', 'B4: SV', 'B5: Drum', 'B6: Air'] i2 __class__ 

[artisan-settings 27052025.txt](https://github.com/user-attachments/files/20450921/artisan-settings.27052025.txt)

commit 1c4e351c (2025-05-27): hardens event name substitution on background load (fixes #1863)
closed 2025-05-27 by MAKOMO