artisan-roaster-scope/artisan #1880: Putting Scale on Burner Slider. And Putting Burner Slider and & Bean Temperature on the same side

feichong83, 2025-06-06:
Dear Artisan,
I have 2 requests which I think if realized,  will make roasting with Artisan more user friendly and efficient.

Problem 1:
1. During the roast, it is usual that burner power is gradually decreased in stages based on the Bean Temperature.
Currently in Artisan, burner slider is displayed on the left while BT is displayed on the right, making it difficult to focus on both parameters at the same time.

Problem 2:
Burner slider does not have scale, making it quite impossible to hit the target % level by just a single slide action. If we have to slide up and down to reach the target level, it may cause temperature swing, resulting in accurate roast curve.

**Describe the solution you'd like**
Request 1:
Move burner slides from left to right side just next to the BT.
Request 2:
Put scale on the slider, example: 100%, 95%, 90%.....

**Describe alternatives you've considered**
Make possible to display burner level in the Main LCD on the right of the screen.

Best Regards and Thanks

roasterdave, 2025-06-06:
Thank you for the suggestions.

#1, Haven't heard this request before.  Not everyone will be happy with such a change so this requires adding another configuration option.  Will mark it as a feature request.
 
#2, Adding a scale is not likely.  The exact slider value is shown in the LCD above it.  When moving the slider, the LCD value updates but the slider value is not registered until you let go of the handle.  Hold until you see the desired value in the LCD then release.  This allows for better accuracy than would be possible with a scale. 

Alternately, if the slider is selected by a click on the handle, you can type the desired value and hit Enter to register it (Esc to cancel).  You can move the selection from one slider to the next with the Tab key.  This allows for rapid and precise positioning of the sliders.

MAKOMO, 2025-06-07:
> **Describe alternatives you've considered** Make possible to display burner level in the Main LCD on the right of the screen.

You didn't reveal which machine setup you are using as in many cases the state represented by the sliders are already read from the machine using extra device configurations with just the corresponding LCDs unticked to hide them to avoid duplicate readings being displayed. In this case you can just toggle the visibility of those LCDs (menu `Config >> Device..`, 2nd tab "Extra Devices").

If not you can add an extra device of type `Slider01` or `Slider23` in the very same dialog to read and report the values from the sliders state while ON.

[image]

[image]

I close this issue as not planned.

closed 2025-06-07 by MAKOMO

feichong83, 2025-06-09:
> <https://github.com/user-attachments/assets/0956bf8c-81ac-4113-a969-05b48b748036> Screenshot.2025-06-07.at.07.36.55.png

Dear Marko,

Thanks for the guidance on how to display the burner power on the main LCD.
I appreciate the prompt response from you and Dave too.

The machine which I am using is a Joper BR15 Advanced. The Burner Power can
be read via MODBUS34 but sometimes there is no reading but only UU shown.
However, I can display it via Slider 23 as well, so no big deal.

As for the request for the location of the Slider to be on the right side,
there are some other reasons for it to be implemented:

1. When we are to move the Burner Slider, it is normal for our eyes to look
left and focus on the Burner Slider, then move the mouse cursor onto the
slider button and then adjust the Burner value. Thus, during this period,
it is not so easy to see the Burner power and Bean temperature in parallel.

2. The above is even more critical during pre first crack to post first
crack period, during when in a split of a second, the ROR curve may flick
or crash if we are slow in adjusting the flame. The time taken to turn our
eyes left, move the cursor to be right on top of the Slider Button, and
then adjust it, this may take up to a few seconds. Nowadays, roasters are
quite determined in getting a smoothly descending ROR curve as it really
has a positive impact on the roast results.

3. In my opinion, the most important and also the most challenging part of
our roasts seem to be the Finishing parts (Pre FC, During FC and Post FC),
whose BT and ROR curve are normally located more to the right of the screen.
Thus, if  the Sliders are located on the right, it is a lot easier for
rosters to be looking at  BT/ BT ROR/ ET ROR  curves and Slider values at
the same time since they are all located at the same side of the screen.

4. For a right handed person, it somehow feels more comfortable to have the
slider to be on the same side with the hand which we use to hold the mouse.

I understand all the above are only my own opinions. Thus it is better for
Artisan to seek the opinions of other professional roasters regarding this
matter before finally deciding on whether to realise this request. Thank
your support.

Best Regards,

Darren

feichong83, 2025-06-11:
> Like this?

Hi Marko,

Yes, like this :)
The BT, ET, BT ROR etc are on the right of the burner slider, as indicated
in the picture below. Thanks for your understanding and support.

[image: Screenshot 2568-06-11 at 17.35.49.png]