artisan-roaster-scope/artisan #1883: Istallation error

beanoccio, 2025-06-08:
### Problem Description

newest continous build win-x64 

[image]

### Artisan Version

3.1.5 Continuous Build

### Computer OS and Version

W11

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [x] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2025-06-11:
The main page states `WARNING: pre-release (continuous) builds may not work. Use at your own risk.`

Maybe we should but this in bold.

closed 2025-06-11 by MAKOMO

beanoccio, 2025-06-11:
Yes, sorry, I know, but it was meant solely as information, not as a complaint.