artisan-roaster-scope/artisan #1931: Issue with MODBUS after update from 3.1.0 to 3.2.0

jpdargent, 2025-07-21:
### Problem Description

Hi there! Thanks for the great work.
I had a problem this morning with my COFFED SR5 communicating with Artisan via MODUS (through USB/Serial).
After updating from 3.1.0 to 3.2.0 I could not reveive Data from the roaster but I could send command: (Airflow and gas were working but I had no reading on BT or ET.
After trying to debug for an hour (changing Serial settings and stuff like that), I had to go back to my previous version to make it work.
Any idea why this happened?
Thanks for the help

### Artisan Version

3.2.0

### Computer OS and Version

Windows 7 Pro SP1 64bits

### Connected devices or roasting machine

Coffed SR5

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [x] No files are attached.

MAKOMO, 2025-08-06:
It turned out that there was a typo which broke MODBUS for some setups on Windows Legacy builds (Windows 7/8) which you are using. This is fixed now. The fix will be part of the next official release and is already available in the [Continuous Builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).

Note that the settings you did attach are somehow a cross-over of some Probat and the Coffed settings. This might cause the MODBUS errors that you observed before already. You can easily fix this by selecting the empty entry from the popup under "Sampling" in the Events dialog, first tab "Config" (menu Config >> Events) marked by the green rectangle in this screenshot.

[image]

MAKOMO, 2025-08-06:
Oh, maybe it is also not a good idea to filter out temperature values higher than 200C. Your decision, but BT and especially ET temperatures are often higher then 200C.

Menu Config >> Curves, 2nd tab "Filters".

[image]

closed 2025-08-06 by MAKOMO

jpdargent, 2025-08-06:
Thank you so much, and ill try leaving the sampling to blank to see if it fixes the error messages.

Regarding BT I never went over 190/193°C on my Coffed SR5, it has enough power to roast fast without having to high temperatures, so I figured I might as well filter data that way

Thanks again!

jpdargent, 2025-08-11:
> Note that the settings you did attach are somehow a cross-over of some Probat and the Coffed settings. This might cause the MODBUS errors that you observed before already. You can easily fix this by [...]

I changed it and got rid of the modbus error message, worked like a charm! Thank you so much!